pf-robotics/kachaka-api #5: DockerHub/Github Container Registoryでimageを公開するgithub actionsの追加

hakuturu583, 2023-08-14:
GRPCはrosdepで依存解決するのが少し難しかったと記憶しており、Dockerで開発ができると非常に嬉しいです。
そこで公開頂いている[こちらのDockerfile](https://github.com/pf-robotics/kachaka-api/blob/main/Dockerfile.ros2)なのですが、docker pullするだけで使えたら非常に嬉しいと考えています。
Github ActionsでDocker ImageをBuildしてPushするのは簡単なのでそれがいいかなと思っています。
イメージを無料で公開できる先として考えられるものに

- Github Container Registory
- Dockerhub

あたりが考えられると思います。
このあたりのサービスを使ってDocker Imageを公開できたらなと思っています。

ただ、リポジトリの内部で[PFRさんのCI設定らしきもの](https://github.com/pf-robotics/kachaka-api/tree/main/.flexci)も見つけてるのでどうコミットしたものかなと考えています。

ご検討よろしくお願いいたします。

youtalk-pfr, 2023-08-14:
早速のご試用ありがとうございます！現在は社内CIツールでイメージを作ってしまっているのですが、イメージ自体は公開しております。
https://github.com/pf-robotics/kachaka-api/blob/main/tools/ros2_bridge/docker-compose.yaml#L5

```
docker pull asia-northeast1-docker.pkg.dev/kachaka-api/docker/kachaka-grpc-ros2-bridge:20230810-x86_64
or
docker pull asia-northeast1-docker.pkg.dev/kachaka-api/docker/kachaka-grpc-ros2-bridge:20230810-aarch64
```
をお試しください。

youtalk-pfr, 2023-08-14:
GitHub Actionsへの移行も計画中です。

hakuturu583, 2023-08-14:
ありがとうございます！確かにPullできました！

youtalk-pfr, 2023-08-14:
移行は検討中ですが、PRいただけると弊社のOSS公開冥利に尽きます！

hakuturu583, 2023-08-14:
片岡は両方普段から利用しておりますのでPR出すことは可能です。
ただdockerhubだとorganizationを作ったりgithubのcontainer registoryだとgithub organizationの設定が必要です。
また無課金だとdocker hubにはrate limitが存在したりします。
そのためPFRさんとしてどちらの方が嬉しいか方針をいただけると作業しやすくなります🙏

hakuturu583, 2023-08-14:
https://github.com/pf-robotics/kachaka-api/issues/5#issuecomment-1676915636

こちらのイメージを以下のコマンドでrunしたところ、ros2コマンドが見つからないようです。
おそらく最初にros_entrypoint.shを実行する想定かと推測するのですがそれが動かなくなっているのかなと思います。
BugFix試みますがとりあえずご報告まで

nozaki-pfr, 2023-08-15:
ありがとうございます。entry_pointで/opt/kachaka/setup.bashを読み込むのが抜けているようです。修正します

closed 2023-09-08 by terakoji-pfr