pf-robotics/kachaka-api #9: 依存関係解決に関する相談

hakuturu583, 2023-08-14:
[この手順に従って](https://github.com/pf-robotics/kachaka-api#%E5%8B%95%E4%BD%9C%E7%A2%BA%E8%AA%8D)ROS2 bridgeの起動を試みたところ、自分の環境にdocker-composeが入っておらず失敗しました。
docker-composeの依存がある旨を記載するPRは出しておきますが、それとは別にpoetryやryeなどによるパッケージ管理を導入してもよいのではないかと感じました。

[docker-coposeはPyPIで公開されている](https://pypi.org/project/docker-compose/)ようですので、モダンなパッケージ管理ツールを用いれば依存関係の解決はより簡単になると思います。

とはいえ自分が勝手にすすめていいことにも思えないので現在requirements.txtで依存解決をされている理由などがありましたら情報いただけると幸いです。

nozaki-pfr, 2023-08-15:
docker-composeは最新のdockerには標準として取り込まれているため、手順に従ってdockerのinstallを行うとdocker composeが実行できる状態かと思います。

poetry等の利用に対しては
* installと利用手順のドキュメンテーションが必要になる
* クロスコンパイルに利用している環境にそれらのサポートを追加する必要がある
という問題があります。
これらに対して得られる利点がrequirements.txtを手動管理しなくて良いあたりだと認識しています。
総合して考えるとpipをそのまま使いたいと考えています。

なお requirements.txt を使うという点には特にこだわりがなく、pythonライブラリのkachaka_apiの依存関係はpyproject.tomlに記述しており、またipynbの依存についてはnotebook中でpipを呼ぶ構成になっています

hakuturu583, 2023-08-17:
ありがとうございます。
poetry等を使うメリットとしては

- 依存をコミットハッシュレベルで簡単に管理できるため、requirements.txtより「Aさんの手元で起きた不具合がBさんの手元では発生しない！」といった不具合が起きにくい(requirements.txtでも同様のことは可能ではあるが、pipのバージョンを揃えたりrequirements.txtにすべてのハッシュを手動で記載する必要がある)
- 仮想環境にインストールしてくれるので手元の環境を汚さない

というものがあります。
メンテナーのPFRさんがpip使い続けたいということでしたら無理に推進することではないと考えますのでpipでいいかなということでしたら、closeいただいて大丈夫です。

closed 2023-08-18 by nozaki-pfr