pf-robotics/kachaka-api #12: /clockトピックの発行

hakuturu583, 2023-08-15:
ROS 2ではclockトピックを用いて分散計算環境の時刻同期を行うと認識していますが、トピックのリストを確認するとそのトピックが出ていないように見えます。

```
/clicked_point
/goal_pose
/initialpose
/kachaka/front_camera/camera_info
/kachaka/front_camera/image_raw
/kachaka/front_camera/image_raw/compressed
/kachaka/imu/imu
/kachaka/layout/locations/list
/kachaka/layout/shelves/list
/kachaka/lidar/scan
/kachaka/manual_control/cmd_vel
/kachaka/mapping/map
/kachaka/mapping/map_updates
/kachaka/object_detection/result
/kachaka/odometry/odometry
/kachaka/robot_info/version
/kachaka_description/joint_states
/kachaka_description/robot_description
/parameter_events
/rosout
/tf
/tf_static
```

rvizで可視化を実行すると以下のようなエラーがでており、使用しているPCが大きく時間がずれているわけではないためNPTによる時刻同期か`clockトピック`による時刻同期が必要かと思います。

```
[INFO] [1692065101.170873319] [rviz]: Message Filter dropping message: frame 'odom' at time 1692065097.799 for reason 'discarding message because the queue is full
```

youtalk-pfr, 2023-08-15:
確かに。次回のアップデートで対応いたします。

youtalk-pfr, 2023-09-13:
`/clock`トピックの発行ではないのですが、 https://github.com/pf-robotics/kachaka-api/pull/39 にてtfに正しくタイムスタンプが振られるようになったので、上記のような時刻ずれの問題はかなり改善されたと思われます。一度ご確認ください。

youtalk-pfr, 2023-09-15:
一旦閉じます。問題が再現されれば、再度開き直します。

closed 2023-09-15 by youtalk-pfr