pf-robotics/kachaka-api #92: カチャカAPI v3.0.14対応

atinfinity, 2024-06-11:
<https://kachaka.zendesk.com/hc/ja/articles/9738402154639-カチャカソフトウェアアップデートのお知らせ-Ver-3-0-14>を読むと、2024/6/11時点でカチャカソフトウェア自体の最新バージョンはv3.0.14となっています。一方でkachaka-apiは[v2.7.8](https://github.com/pf-robotics/kachaka-api/releases/tag/v2.7.8)までの対応となっています。v3.0.14のAPI対応予定をお教え頂きたいです。

hidai-pfr, 2024-06-11:
3.0.14 対応の tag は付与できていない状況ですが、main ブランチでは https://github.com/pf-robotics/kachaka-api/commit/60280eda3caafa7a133c814f62d516e92da36b55 で gRPC API レベルでの 3.0.14 対応がされています。
その後いくつか commit が積まれていて、Python の kachaka_api ライブラリの対応と ROS 2 ブリッジの対応になりますが、Python と ROS 2 に関してはすべての機能対応が追いついていない状況ではあり、順次対応予定です。

もし具体的な要望があればコメントして頂ければ対応を検討します。

atinfinity, 2024-06-14:
@hidai-pfr コメントありがとうございます。状況について理解できました。<https://github.com/pf-robotics/kachaka-api/commit/60280eda3caafa7a133c814f62d516e92da36b55>にある機能ですと、

- バッテリー情報の取得
- ToFセンサ画像/情報の取得
- フロント・バックLEDライトの点灯

Pythonのkachaka_apiライブラリの対応、ROS 2ブリッジ対応が特に気になっております。

atinfinity, 2024-07-26:
https://github.com/pf-robotics/kachaka-api/releases/tag/v3.0.14 で対応されているようなので閉じます。

closed 2024-07-26 by atinfinity