pf-robotics/kachaka-api #100: ros2_bridgeについて

atinfinity, 2024-07-04:
ros2_bridgeを使うために<https://github.com/pf-robotics/kachaka-api?tab=readme-ov-file#動作確認>を読んでいます。READMEにある

```bash
cd ~/kachaka-api/tools/ros2_bridge
./start_bridge.sh <カチャカのIPアドレス>
```

では、Dockerイメージをpullし、Dockerコンテナを起動して、ros2_bridgeを実行しているという理解です。ここでpullされるDockerイメージはx86_64、arm64どちらのアーキテクチャ向けも提供していますか？arm64のホスト環境で実行しようと思っています。

atinfinity, 2024-07-05:
arm64マシン上で試したところ、docker pullして試したところamd64前提のようでした。

```bash
docker pull asia-northeast1-docker.pkg.dev/kachaka-api/docker/kachaka-grpc-ros2-bridge:20240519
```

```bash
docker run -it asia-northeast1-docker.pkg.dev/kachaka-api/docker/kachaka-grpc-ros2-bridge:20240519 bash
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
exec /usr/bin/dumb-init: exec format error
```

DockerイメージのもとになっているDockerfileは<https://github.com/pf-robotics/kachaka-api/blob/main/Dockerfile.ros2>だと想像しているのですが、本ファイルは1年以上更新がなさそうであるため、最新のものに追従されていない可能性もありそうだと考えています。可能であればコメント頂きたいです。

terakoji-pfr, 2024-07-05:
こちら、配布しているdocker imageはx86_64のみです。
Dockerfile自体は最新なのでご指摘のdockerファイルでarm64でbuildしていただければ大丈夫です。

atinfinity, 2024-07-07:
@terakoji-pfr 回答ありがとうございました。ROS 2連携でarm64環境（RasPi、Jetsonなど）が使われることも考えられるのでarm64向けのDockerイメージ提供やREADMEでのインストラクション追記などもご検討頂けると助かります。

atinfinity, 2024-07-08:
@terakoji-pfr 頂いた情報をもとにROS 2ブリッジを試していますが正常に動作しませんでした。

## 現象

ROS 2ブリッジが正常に起動しない。

## 環境

- カチャカ：3.0.14
- kachaka-api
  - https://github.com/pf-robotics/kachaka-api/commit/28540d353502e23b6699c14ad15610b7875faca7
- ROS 2実行環境
  - Arm64
  - ROS 2ブリッジはDockerコンテナで起動

## ログ

以下のエラーが出て、ROS 2ブリッジが正常に起動しません。

> ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.215949281] [kachaka.grpc_ros2_bridge_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::TofCameraComponent>
ros2_bridge-1  | [component_container_mt-1] terminate called after throwing an instance of 'cv::Exception'
ros2_bridge-1  | [component_container_mt-1]   what():  OpenCV(4.5.4) ./modules/imgcodecs/src/loadsave.cpp:814: error: (-215:Assertion failed) !buf.empty() in function 'imdecode_'

```shell
cd ~/kachaka-api/tools/ros2_bridge
./start_bridge.sh <カチャカのIPアドレス>

Attaching to ros2_bridge-1
ros2_bridge-1  | [INFO] [launch]: All log files can be found below /tmp/2024-07-08-04-12-50-029482-jetson-orin-nx01-3785
ros2_bridge-1  | [INFO] [launch]: Default logging verbosity is set to INFO
ros2_bridge-1  | [INFO] [component_container_mt-1]: process started with pid [3842]
ros2_bridge-1  | [INFO] [robot_state_publisher-2]: process started with pid [3844]
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.516823425] [kachaka_description.robot_state_publisher]: got segment base_footprint
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517023425] [kachaka_description.robot_state_publisher]: got segment base_l_drive_wheel_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517045345] [kachaka_description.robot_state_publisher]: got segment base_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517055073] [kachaka_description.robot_state_publisher]: got segment base_r_drive_wheel_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517062849] [kachaka_description.robot_state_publisher]: got segment camera_back_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517069985] [kachaka_description.robot_state_publisher]: got segment camera_front_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517077217] [kachaka_description.robot_state_publisher]: got segment docking_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517084129] [kachaka_description.robot_state_publisher]: got segment imu_link
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517091201] [kachaka_description.robot_state_publisher]: got segment laser_frame
ros2_bridge-1  | [robot_state_publisher-2] [INFO] [1720411970.517097889] [kachaka_description.robot_state_publisher]: got segment tof_link
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.647717569] [kachaka.grpc_ros2_bridge_container]: Load Library: /opt/kachaka/lib/libkachaka_grpc_ros2_bridge.so
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.978428737] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::AutoHomingComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.978553057] [kachaka.grpc_ros2_bridge_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::AutoHomingComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.990088705] [grpc_ros2_bridge]: created stub 10.5.244.5:26400
ros2_bridge-1  | [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/kachaka/auto_homing' in container '/kachaka/grpc_ros2_bridge_container'
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.997218177] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::AutoHomingComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.997312865] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::BackCameraComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411970.997324065] [kachaka.grpc_ros2_bridge_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::BackCameraComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.008875777] [kachaka.back_camera]: 'back_camera' bridge starting
ros2_bridge-1  | [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/kachaka/back_camera' in container '/kachaka/grpc_ros2_bridge_container'
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.009886945] [kachaka.back_camera]: 'back_camera' bridge starting
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.015111969] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::AutoHomingComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.015214209] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::BackCameraComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.015227521] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::DynamicTfComponent>
ros2_bridge-1  | [component_container_mt-1] [INFO] [1720411971.015237025] [kachaka.grpc_ros2_bridge_container]: Found class: rclcpp_components::NodeFactoryTemplate<kachaka::grpc_ros2_bridge::FrontCameraComp [... cut ...]
ros2_bridge-1  | [ERROR] [component_container_mt-1]: process has died [pid 3842, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=grpc_ros2_bridge_container -r __ns:=/kachaka'].
```

```shell
$ ros2 topic list
/kachaka_description/joint_states
/kachaka_description/robot_description
/parameter_events
/rosout
/tf
/tf_static
```

## 調査

<https://github.com/pf-robotics/kachaka-api?tab=readme-ov-file#サンプルコードをjupyterlabで実行する>にあるNotebookファイルでカメラ画像、LiDARがキャプチャできていること確認済。

## 疑問点

- PFRでarm64の系は動作確認済でしょうか？
- カチャカセットアップ時にカチャカのソフトウェアアップデートが促されたので3.0.14にアップデートしましたが、この状態だとROS 2ブリッジが動作しないのでしょうか？

terakoji-pfr, 2024-07-08:
1. arm64でのbridgeは動作確認しておりませんので、検証後お伝えします。
2. アップデートは無関係です。
 
arm系の検証も行っていくので少々お待ちいただければと思います。

atinfinity, 2024-07-09:
@terakoji-pfr 

> ros2_bridge-1 | [component_container_mt-1] what(): OpenCV(4.5.4) ./modules/imgcodecs/src/loadsave.cpp:814: error: (-215:Assertion failed) !buf.empty() in function 'imdecode_'

のメッセージを手掛かりにコードを確認しました。
<https://github.com/pf-robotics/kachaka-api/blob/main/ros2/kachaka_grpc_ros2_bridge/src/component/mapping_component.cpp>

おそらく、このコンポーネントではカチャカ内に地図データが既にあることを前提にされているのではないかと考えています。ROS 2で使う際、mapping、localizationは

atinfinity, 2024-07-09:
@terakoji-pfr カチャカが一定時間静止していると、scanトピックの配信がされなくなる挙動があります。ROS 2で使う場合、望ましくない挙動なのですが抑制する方法はありますか？

terakoji-pfr, 2024-07-09:
scanトピックを抑制する方法ですが、
https://github.com/pf-robotics/kachaka-api/blob/main/python/demos/get_laser_scan.ipynb
に記載されているactivatorを使うとトピックが止まらないようになります。

ros側にinterfaceを現状では出していないので、kachaka-apiと併用してお使いください。

terakoji-pfr, 2024-07-09:
@atinfinity ご指摘の通りmapがない場合には、arm64, aarch64問わずに

```
ros2_bridge_1            | [component_container_mt-1] terminate called after throwing an instance of 'cv::Exception'
ros2_bridge_1            | [component_container_mt-1]   what():  OpenCV(4.5.4) ./modules/imgcodecs/src/loadsave.cpp:814: error: (-215:Assertion failed) !buf.empty() in function 'imdecode_'
```
のエラーが出ることを確認しました。
このエラーは今後のリリースで修正します。

ワークアラウンドとして当面は最低一つmapを作りremapする等でその他のアルゴリズムを試していただけないでしょうか？
こちらは別の方が試していて確認取れております。

atinfinity, 2024-07-10:
@terakoji-pfr ご回答ありがとうございました。現状について理解できました。次バージョンまではご提案頂いたワークアラウンドを行うことにします。

terakoji-pfr, 2024-07-10:
関連したチケットを切り、このIssueはcloseします。

closed 2024-07-10 by terakoji-pfr