pf-robotics/kachaka-api #103: カチャカが静止していてもscan配信を継続したい

atinfinity, 2024-07-16:
<https://github.com/pf-robotics/kachaka-api/issues/100#issuecomment-2217232092>で質問した件です。<https://github.com/pf-robotics/kachaka-api/issues/100#issuecomment-2217399096>で

> scanトピックを抑制しない方法ですが、
https://github.com/pf-robotics/kachaka-api/blob/main/python/demos/get_laser_scan.ipynb
に記載されているactivatorを使うとトピックが止まらないようになります。
>
> ros2側にinterfaceを現状では出していないので、kachaka-apiと併用してお使いください。

というご回答を受けたので、[python/demos/get_laser_scan.ipynb](https://github.com/pf-robotics/kachaka-api/blob/main/python/demos/get_laser_scan.ipynb)を参考にして、以下のコードをNotebook上で実行していますが、カチャカが10秒静止しているとscanの配信が止まってしまいました。

```python
import kachaka_api
from kachaka_api.util.vision import LaserScanActivator

client = kachaka_api.aio.KachakaApiClient()

async def get_and_show_laser_scan_loop():
    client.ros_laser_scan.stream()

activator = LaserScanActivator()
with activator.activate():
    await get_and_show_laser_scan_loop()
```

Notebookのコメントに

> 上記の方法は、手動操縦モードをONにすることでLiDARを動かしており、カチャカが動かずしばらく経つとスキャンが止まります。

とあるので、`client.set_manual_control_enabled(True)`は呼び出していません。Python APIの呼び出し方が意図通りになっていませんか？

atinfinity, 2024-07-18:
@terakoji-pfr 本件についてご見解頂けないでしょうか？

terakoji-pfr, 2024-07-19:
@atinfinity 
上部のclient.ros_laser_scan.stream()ブロックされています。

最小限のコードは以下のようになります。
`import kachaka_api
from kachaka_api.util.vision import LaserScanActivator
import asyncio

client = kachaka_api.aio.KachakaApiClient()

activator = LaserScanActivator()
with activator.activate():
    while True:
        await asyncio.sleep(1)`

kmurase-pfr, 2024-07-19:
@atinfinity 
async generatorを生成しているだけで、activatorのところを抜けてしまっていると思います。
例えば、元のコードで以下のように最後にprintを入れると、"activator finished"と表示されてしまうと思います。

```python
import kachaka_api
from kachaka_api.util.vision import LaserScanActivator

client = kachaka_api.aio.KachakaApiClient()

async def get_and_show_laser_scan_loop():
    client.ros_laser_scan.stream()

activator = LaserScanActivator()
with activator.activate():
    await get_and_show_laser_scan_loop()

print ("activator finished")
```

ちゃんとasync generatorを使い中の処理を書いてあげると(とりあえず以下は何もしていないですが)、activatorが動き続けてLiDARが止まることはないはずです。

```python
import kachaka_api
from kachaka_api.util.vision import LaserScanActivator

client = kachaka_api.aio.KachakaApiClient()

async def get_and_show_laser_scan_loop():
    async for scan in client.ros_laser_scan.stream():
        # do something
        pass

activator = LaserScanActivator()
with activator.activate():
    await get_and_show_laser_scan_loop()

print ("activator finished")
```

atinfinity, 2024-07-19:
@terakoji-pfr ありがとうございます。動作確認致します。

atinfinity, 2024-07-22:
@terakoji-pfr 頂いたコードで意図通りの挙動（＝カチャカがしばらく静止していてもscanキャプチャが継続される）になること確認致しました。

closed 2024-07-22 by atinfinity