pf-robotics/kachaka-api #105: ROS 2ブリッジから地図、自己位置を配信しない

atinfinity, 2024-07-18:
<https://qiita.com/terakoji-pfr/items/0f1535b45fda58edad83>に以下の記述があります。

> この記事では、マップの管理と位置の推定はkachakaが持っているものを利用したほうが簡単なのでそのまま使い、それ以外のコストマップを作る部分やパスプラン、制御の部分をnavigation2から使っています。

「マップの管理と位置の推定はkachakaが持っているものを利用したほうが簡単」とあるので、一見するとROS 2ブリッジを使って得られるセンサデータ（scanなど）を用いて、自身のmappingを試すこともできそうに読み取れなくもないのですが、現状のROS 2ブリッジからはマップデータおよびodom、map間のtfが出ているため、自身のmappingアルゴリズムを試せないように思っています。

ROS 2ブリッジから地図、自己位置を配信しないようにすることについて今後対応の予定はありますか？

terakoji-pfr, 2024-07-19:
こちら問題をクリアにしたいのですが
/map を共通にし、　/odomを別名とすることで解決できないでしょうか？

それとも
/odom -> /base_footprint
が再利用できないことがクリティカルな問題でしょうか？
こちらは/tf_prefixを用い、 odomトピック(tfでなく)から出すということで解決できますでしょうか？

terakoji-pfr, 2024-07-19:
少し調べましたが、以下で特に今のros2_bridgeがだしているtf, topic等を止めなくても自身のmapping、localizationアルゴリズムを動かせそうです。

1. 自身のmapping、localizationが発行する /odomのtfの名前を /{pfrefix}_odom にする
2. /{prefix}_odom -> /{prefix}_base_footprint のtfを /odom or /odom_rawトピックから作って発行する
3. robot_state_publisherを frame_pfrefixに {prefix}を設定して実行しておく

https://github.com/ros/robot_state_publisher/blob/8b5af4f31e0f754c7c262eb487ccd3956cb788cc/src/robot_state_publisher.cpp#L124

atinfinity, 2024-07-19:
@terakoji-pfr ありがとうございます。試してみます。

atinfinity, 2024-09-10:
<https://zenn.dev/ame_b/articles/da4636a0a8048f>を参考にし、カチャカROS 2ブリッジが配信する地図は利用し、カチャカROS 2ブリッジからodom、map間のtfを配信しないようにすることで対応できました。ROS 2ブリッジの起動オプションで切り替えられるとよいかもしれません。

closed 2024-09-10 by atinfinity