pf-robotics/kachaka-api #106: kachaka_nav2_bringupパッケージのナビゲーション時にエラーが出る

atinfinity, 2024-07-24:
# 現象

kachaka_nav2_bringupパッケージのnavigation_launch.py実行時に以下のエラーが発生し、走行ががたつくことがあります。map、odom間の変換が正常にできないことがあるようです。

```
[controller_server-1] [INFO] [1721806791.956888899] [controller_server]: Passing new path to controller.
[controller_server-1] [ERROR] [1721806792.114764736] [tf_help]: Transform data too old when converting from odom to map
[controller_server-1] [ERROR] [1721806792.114836098] [tf_help]: Data time: 1721806792s 88418722ns, Transform time: 1721806791s 859294000ns
```

# 前提

- カチャカアプリで地図を作成し、地図上の自己位置もカチャカのものを使っています
- カチャカROS 2ブリッジを動かしています
- ナビゲーションはリポジトリ内のサンプルを使用しています
  - <https://github.com/pf-robotics/kachaka-api/blob/main/ros2/demos/kachaka_nav2_bringup/launch/navigation_launch.py>

# 調査内容

map、odom間のtfはkachaka_grpc_ros2_bridgeが出しています。このtfのタイムスタンプ設定処理を読むとgRPCで渡されたタイムスタンプをそのまま付けているように見受けられます。
<https://github.com/pf-robotics/kachaka-api/blob/main/ros2/kachaka_grpc_ros2_bridge/src/converter/ros_header.cpp>

そのため、カチャカ内部の時刻とROS 2を動かすPCの時刻がずれていることが原因なのではないかと考えていますが、見解を頂きたいです。

terakoji-pfr, 2024-07-24:
grpcのprotocolに付与している時間はカチャカ内部の時間です。
ここはそのタイムスタンプを渡さないと他のセンサとの関係もあり、破綻してしまいます。

ご指摘の通りPCとカチャカの時間がずれている可能性があります。
PC側とカチャカのdelayを調べるために,ros2bridgeが上がった状態で

`ros2  topic delay /kachaka/imu/imu`

等でPCとdelayが大きくないかみてもらえないでしょうか？

atinfinity, 2024-07-24:
@terakoji-pfr コメントありがとうございます。

> PCとdelayが大きくないかみてもらえないでしょうか？

結果は以下の通りです。

```shell
$ ros2 topic delay /kachaka/imu/imu
average delay: 0.018
min: 0.011s max: 0.028s std dev: 0.00491s window: 10
average delay: 0.017
min: 0.009s max: 0.028s std dev: 0.00522s window: 20
average delay: 0.016
min: 0.008s max: 0.028s std dev: 0.00495s window: 29
average delay: 0.016
min: 0.008s max: 0.038s std dev: 0.00626s window: 39
average delay: 0.016
min: 0.008s max: 0.038s std dev: 0.00582s window: 49
average delay: 0.016
min: 0.008s max: 0.038s std dev: 0.00562s window: 59
```

atinfinity, 2024-07-24:
記載漏れておりましたが、カチャカROS 2ブリッジ、ナビゲーションを動作させているUbuntu PCはインターネット接続により時刻合わせをしています。NTPサーバーはUbuntu初期設定から変更していません。

terakoji-pfr, 2024-07-24:
問題のあるdelayではなさそうですね。

同様のエラーがnav2でいくつか報告されているようなのでnavigation2側の問題かもしれません。

https://github.com/ros-navigation/navigation2_tutorials/issues/25
https://answers.ros.org/question/344688/transform-data-too-old-when-converting-from-odom-to-map/

terakoji-pfr, 2024-07-24:
ひとまず

https://github.com/ros-navigation/navigation2/blob/d1ad6401fa8939a8eee6c6d15b31cd656995ecb5/nav2_costmap_2d/src/costmap_2d_ros.cpp#L130C26-L130C30

にあるtoleranceでかかっていそう（逆に他では遅れていなさそう）なのでこちらを変更してみて試していただくことできますか？

atinfinity, 2024-07-24:
お返事ありがとうございます。controller_serverでエラーが出ていたので、ご回答頂く前にkachaka_nav2_bringup/params/nav2_params.yamlにあるFollowPathのtransform_toleranceを0.2から0.5に変更するとこのエラーが抑制できていました。

atinfinity, 2024-07-30:
https://github.com/pf-robotics/kachaka-api/pull/110 がマージされたのでクローズします。

closed 2024-07-30 by atinfinity