pf-robotics/kachaka-api #114: [LiDAR]  Scan data publishing is stopped when robot not moving

MrKeith99, 2024-08-10:
[ENG]
The LiDAR sensor is exhibiting inconsistent behavior regarding data publication. Specifically:

* **Data publication starts:** Only when the robot begins to move.
* **Data publication stops:** Approximately 5 seconds after the robot comes to a complete stop.

This behavior is undesirable as it limits the ability to utilize LiDAR data for static environment mapping and obstacle detection during stationary periods.

To address this issue, it would be desirable to introduce a ROS service that can toggle the LiDAR scan data publication on and off independently of the robot's movement state.

---

[日本語]
LiDARセンサーからのデータはパブリッシュされないときがあります。具体的には以下の通りです。

* **データパブリッシュ開始:** ロボットが動き始めた時。
* **データパブリッシュ停止:** ロボットが完全に停止してから約5秒後。

静止状態での環境マッピングや障害物検知にLiDARデータを利用することを制限されます。

この問題に対処するために、ロボットの移動状態に関係なくLiDARスキャンデータのパブリッシュをオン/オフできるようにROSサービスを導入することのよって解決できるかと考えられます。

**Additional Information:**
[image]

terakoji-pfr, 2024-08-13:
#103

MrKeith99, 2024-08-23:
@terakoji-pfr 
Thank you for your feedback.
We will have a look on it.
Meanwhile, it looks like it is possible to obtain the sensor data after Kachaka stops, so I will close this issue.
Thank you!

closed 2024-08-23 by MrKeith99