pf-robotics/kachaka-api #115: [LiDAR] Scan Data lenght is not constant and header might be incorrect

MrKeith99, 2024-08-10:
[ENG]
We are experiencing issues with obtaining a correct scan data from the LiDAR which is being published on the `/kachaka/lidar/scan` topic.

**Scan Data Header content:**
* The published scan data indicates a full 360-degree coverage (`angle_min: -3.1415927410125732`, `angle_max: 3.1415927410125732`).
* However, physical observations of the robot suggest an occlusion in the rear field of view.
* This discrepancy implies that the published header information (e.g., angle_min, angle_max) might be incorrect.

**Scan Data Length Variation:**
* The length of the `ranges` array in the scan data is not constant across scans.
* This inconsistency suggests potential data merging, possibly with data from another sensor like a TOF sensor.

It would be beneficial to access the raw LiDAR data and any additional sensor data (e.g., TOF) independently.

---

[日本語]
`/kachaka/lidar/scan` トピックでパブリッシュされているLiDARのスキャンデータに不整合が見られます。

**スキャンデータヘッダの内容:**
* 公開されているスキャンデータは360度の全方位をカバーしていることを示しています (`angle_min: -3.1415927410125732`, `angle_max: 3.1415927410125732`)。
* しかし、実機を確認したところ、後方の視野が遮られていることがわかりました。
* この不一致から、パブリッシュされているヘッダ情報 (angle_min, angle_max など) が正しくない可能性があります。

**スキャンデータの変動:**
* スキャンデータの `ranges` 配列の長さがスキャンごとに変わっています。
* 他のセンサー (TOFセンサーなど) のデータとマージされているからだと考えられます。

生のLiDARデータと他のセンサーデータ (TOFなど) に個別にアクセスできる方法があれば、ご教示ください。

### Additional Information:
Here we share the information got with `ros2 topic echo /kachaka/lidar/scan`:
```
header:
  stamp:
    sec: 1723212284
    nanosec: 860031000
  frame_id: laser_frame
angle_min: -3.1415927410125732
angle_max: 3.1415927410125732
angle_increment: 0.00998916570097208
time_increment: 0.00019924482330679893
scan_time: 0.12532499432563782
range_min: 0.10000000149011612
range_max: 10.0
ranges:
- 0.0
- 5.0329999923706055
- 5.044000148773193
- 0.0
- 5.057000160217285
- 5.071000099182129
- ...
```

Where `ranges` length over the scans seems to be similar to this:
```
[INFO] [1723212225.861439148] [lidar_length_node]: Length of LiDAR ranges: 646
[INFO] [1723212225.959220199] [lidar_length_node]: Length of LiDAR ranges: 661
[INFO] [1723212226.079582270] [lidar_length_node]: Length of LiDAR ranges: 669
[INFO] [1723212226.188875728] [lidar_length_node]: Length of LiDAR ranges: 672
[INFO] [1723212226.406561765] [lidar_length_node]: Length of LiDAR ranges: 665
[INFO] [1723212226.527574158] [lidar_length_node]: Length of LiDAR ranges: 659
[INFO] [1723212226.631671547] [lidar_length_node]: Length of LiDAR ranges: 652
[INFO] [1723212226.741832658] [lidar_length_node]: Length of LiDAR ranges: 645
[INFO] [1723212226.862682051] [lidar_length_node]: Length of LiDAR ranges: 636
[INFO] [1723212226.982428452] [lidar_length_node]: Length of LiDAR ranges: 634
...
```

terakoji-pfr, 2024-08-13:
#103

terakoji-pfr, 2024-08-13:
With kachaka, rearward scan data cannot be acquired because the LiDAR is blocked by the docking unit.

Also, the number of scan points varies depending on the rotation speed of the motor.

MrKeith99, 2024-08-23:
@terakoji-pfr 
Thank you for your prompt replay.

> With kachaka, rearward scan data cannot be acquired because the LiDAR is blocked by the docking unit.

We could imagine that it was because of the docking unit.
However, is it possible to know the actual detecting range of the LiDAR,
which I believe is not `angle_min: -3.1415927410125732`, `angle_max: 3.1415927410125732`.

> Also, the number of scan points varies depending on the rotation speed of the motor.

I see! We have tried with other LiDAR sensors, and it was the first time to observe this phenomenon.
I will have a look how we can process the data so that we can obtain a constant number of points for processing.

atinfinity, 2024-09-11:
@terakoji-pfr LaserScanのmsg仕様におけるangle_min、angle_maxについては<https://docs.ros2.org/latest/api/sensor_msgs/msg/LaserScan.html>に情報があります。

もし、カチャカの2D LiDARが前方しか見ていないとしたら、現状のカチャカROS 2ブリッジで配信されるscanトピックで設定されているangle_min、angle_maxは360度見ていることになっているので、実状と即していないのではないかという質問だと思います。一度ご確認頂きたいです。

terakoji-pfr, 2024-12-01:
iHR研究会で直接お話できましたが、ここにも書いておきます。

LiDARの所得点数は、kachakaのLiDARでは変動します。
固定長のデータが必要でしたらresampling等で対応お願いします。

LiDARの隠れは

[image]

となっていますが、normalの場合LIDARの方式による成約からこのカバーより数[deg]見えない範囲があります。

terakoji-pfr, 2024-12-01:
angle_min, angle_maxはカバーでなくてセンサーとしての仕様なので取れるポイントとは異なる値になります。

atinfinity, 2024-12-01:
@terakoji-pfr コメントありがとうございます。点数が変動することは理解できました。一方、sensor_msgs/msg/LaserScanにて特定の範囲内（例えば正面±30度）のスキャンだけ使いたいときにインデックスで参照するケースがあると思いますが、点数が変動するということは同じインデックスでも見ている角度が異なることになります。そのため、カチャカのLiDARデータを使ったソフトウェア開発では厳密な範囲指定による参照ができないという理解でよいでしょうか？

terakoji-pfr, 2024-12-02:
@atinfinity 点数は変動するのですが、点の角度は

angle = angle_min + angle_increment * index 

なので厳密な範囲指定による参照はできます。
angle_incrementがモーターの回転により変動してしまうので、厳密に同じ角度のデータが必要な場合は上述しているようにresamplingして補間するくらいしか無いです。

このあたりの変動はproとnormalで少し違いがあり、proのほうがangle_incrementの変動幅は少ないです。

atinfinity, 2024-12-02:
@terakoji-pfr コメントありがとうございます。angle_incrementが動的に変わる旨、理解できました。

terakoji-pfr, 2024-12-05:
こちらは解決したと思いますので閉じさせていただきます。

closed 2024-12-05 by terakoji-pfr