pf-robotics/kachaka-api #116: 充電機付近へMoveToPoseCommandを実行した場合に、RUNNING状態のままとなる

yutaroha, 2024-08-22:
カチャカのgRPC APIを使用し、MoveToPoseCommandで移動先を充電器から近い位置(充電器前方 100mmあたり)を指定して実行すると、GetCommandStateの結果COMMAND_STATE_RUNNINGのままとなります。(カチャカのスマホアプリで見ても実行中のまま)

kachaka-api/python/demos/plot_map_robot_lidar.ipynbのクリックした位置にカチャカを移動させるデモを使用して、充電器付近をクリックしても同様の現象が確認できました。

こちらでの再現性は数回試して、毎回発生しているようでした。

ご確認いただけますと幸いです。

hidai-pfr, 2024-08-22:
こちらで再現できなかったので、いくつか確認させてください。
指示した目的地にはたどり着いているのでしょうか？
たどり着いているとしたら、カチャカは完全に停止していますか？
たどり着いていないとしたら、カチャカの状態はどのようになっていますか？

hidai-pfr, 2024-08-22:
いや、再現しました。確認します。

kmurase-pfr, 2024-08-23:
MoveToPoseCommandを使用した際は、
充電器や壁など、障害物にかなり接近している地点への移動の場合、
到達判定周りの挙動に問題があることが分かりました。

今後のリリースで修正を検討予定ですが、
現状はMoveToLocationやMoveShelf、MoveForwardなどのコマンドで代用して頂くか、
GetRobotPoseのAPIを利用して自前で到達判定を行い、移動キャンセルを投げることで対応頂けないでしょうか。

yutaroha, 2024-08-26:
承知いたしました。ご回答いただきありがとうございます。

kmurase-pfr, 2024-12-04:
障害物が近くにあって、ゴールに辿り着けない場合に、不適切に終了判定が行われてカチャカが最後の旋回を行ってしまう、という挙動自体は最新リリースでは修正が入っています。

ただ、そういうぎりぎりの地点への移動の場合にゴール到着判定を緩めたいなどの調整が行えるようなAPIがあるとよいので、今後サポートを検討します。

本チケットはクローズします。

closed 2024-12-04 by kmurase-pfr