pf-robotics/kachaka-api #117: カチャカ内蔵の自己位置推定機能の位置ズレ修正

atinfinity, 2024-08-23:
現在、カチャカ内蔵の自己位置推定機能を使っていますが、カチャカの2D LiDARが前方しか見ていないためか、RViz2で見ていると自己位置推定結果がズレていることがよく見受けられます。以下の点について教えてください。

- カチャカ内蔵の自己位置推定でロストしていそうか、もしくは信頼度を得る方法はありますか？
- <https://kachaka.zendesk.com/hc/ja/articles/6032866920207-ボイスコマンド一覧>の「ねぇカチャカ、落ち着いて」相当のことをカチャカAPI、もしくはROS 2で実行できますか？

terakoji-pfr, 2024-08-23:
フィードバックありがとうございます。

自己位置推定の結果の信頼度は現状では取得する方法ありません。

落ち着いて相当のことは、今のAPIでは出していません。
今後のリリースでチームで検討させていただきます。

atinfinity, 2024-08-26:
@terakoji-pfr ご回答ありがとうございました。

> 今後のリリースでチームで検討させていただきます。

検討お願いします。また、仮に「ねぇカチャカ、落ち着いて」相当のことができたとして、これをどのタイミングで呼ぶべきかという話は残ってしまうと思われるので、

> カチャカ内蔵の自己位置推定でロストしていそうか、もしくは信頼度を得る方法

についてもご検討頂きたいです。

terakoji-pfr, 2024-12-10:
こちら自己位置の信頼度を進める予定ですが、

現状
https://github.com/pf-robotics/kachaka-api/blob/main/python/demos/robot_pose.ipynb
で自己位置を修正する方法があるのでこちらもご検討ください。

atinfinity, 2024-12-10:
@terakoji-pfr 情報ありがとうございます。`client.set_robot_pose()`を呼ぶと、その際に「ねぇカチャカ、落ち着いて」相当の補正処理が動作するという理解で合っていますでしょうか？

terakoji-pfr, 2024-12-10:
いえ、こちらは類推の通りinitial_pose相当です。
落ち着いて相当は進めてますのでお待ち下さい

hidai-pfr, 2025-01-31:
> 「ねぇカチャカ、落ち着いて」相当のことをカチャカAPI、もしくはROS 2で実行できますか？

これに関しては、3.8 で対応しました。

closed 2025-01-31 by hidai-pfr