pf-robotics/kachaka-api #120: 配布dockerイメージ以外ではmapトピックが正しく配信されない

yoshikai-qt, 2024-09-06:
/kachaka/mapping/mapについて、配布dockerイメージ(asia-northeast1-docker.pkg.dev/kachaka-api/docker/kachaka-grpc-ros2-bridge)以外では正しく配布されないようです。

問題に気がついた経緯ですが、kachaka_grpc_ros2_bridgeに変更を加えたものを検証していて、READMEの「Dockerイメージのビルド」に書かれているやり方で、イメージをビルドし、kachaka-api:latestを使うようにtools/ros2_bridge/docker-compose.yaml を書き換えたところ、追加した機能は反映されるようになった一方で、mapトピックの配信がされなくなりました。

dockerのビルドの問題かどうかを切り分けるために念の為、ネイティブビルドして直接launchファイルを叩くアプローチも試してみましたが状況は同じでした（画像など他のトピックは正常に配信されているがmapトピックは出力されない）。

何か配布dockerイメージのみに含まれている修正等あるのでしょうか。どの時点から配信されなくなったのか定かではないのですが、少なくとも2.7.8まではネイティブビルドで期待通りmapトピックの配信はできていたように記憶しています。

どうぞ、ご確認よろしくお願い致します。

terakoji-pfr, 2024-09-09:
ご迷惑お消します。今配信されていないバージョンを教えていただけますでしょうか？

yoshikai-qt, 2024-09-10:
@terakoji-pfr お返事ありがとうございます。最新版のv3.3.7にて確認をしております。どうぞ、よろしくお願い致します。

yoshikai-qt, 2024-10-25:
@terakoji-pfr
こちらについて、バグの原因箇所を特定できたので、修正してみました。
以下のPRで投げましたのでご確認頂ければ幸いです。最新版のv3.4.7で確認修正しています。
https://github.com/pf-robotics/kachaka-api/pull/126

yoshikai-qt, 2024-10-28:
@terakoji-pfr PRのご確認、マージありがとうございました。
こちらのissueは解決されたということでcloseさせていただきます。
どうぞ、よろしくお願い致します。

closed 2024-10-28 by yoshikai-qt