pf-robotics/kachaka-api #159: kachaka-apiパッケージがインストールできない

atinfinity, 2025-02-07:
- Ubuntu 22.04
- Python 3.10
- x86_64, aarch64共通

<https://github.com/pf-robotics/kachaka-api/blob/v3.8.5/docs/PYTHON.md>にある手順でkachaka-apiパッケージのインストールを試みています。

```bash
pip install --extra-index-url https://pf-robotics.github.io/kachaka-python-packages/simple kachaka-api
```

上記コマンド実行後、以下のエラーが出てインストールできませんでした。

```bash
ERROR: Could not find a version that satisfies the requirement kachaka-api (from versions: 1.0.0, 2.3.8, 2.5.7, 2.7.8, 3.0.14, 3.1.10, 3.2.6, 3.3.7, 3.4.7, 3.6.6, 3.6.6.1, 3.7.6, 3.8.5)
ERROR: No matching distribution found for kachaka-api
```

<https://github.com/pf-robotics/kachaka-api/blob/v3.8.5/README.md>を読むと、

> Python 3.10+

とあり、READMEの説明を読むと要件は満たしているように思います。そのため本件について確認頂きたいです。また、ソースコードからのインストール手順についても情報あると助かります。

## 追加調査

- Ubuntu 24.04
- Python 3.12

の環境では

```bash
pip install --extra-index-url https://pf-robotics.github.io/kachaka-python-packages/simple kachaka-api
```

で問題なくインストールできました。

atinfinity, 2025-02-07:
- Ubuntu 22.04
- Python 3.10

にてソースから入れることを試みましたが、こちらも正常にインストールできていません。

```bash
git clone https://github.com/pf-robotics/kachaka-api.git -b v3.8.5
cd kachaka-api
pip install .
Defaulting to user installation because normal site-packages is not writeable
Processing /home/humble/kachaka-api
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: UNKNOWN
  Building wheel for UNKNOWN (pyproject.toml) ... done
  Created wheel for UNKNOWN: filename=UNKNOWN-0.0.0-py3-none-any.whl size=5114 sha256=fe1d81d6dead0d484255d1acc32c13a946b0d8d897a3d29e2205a674e4ff98c2
  Stored in directory: /home/humble/.cache/pip/wheels/d1/c5/20/3a62c7db44a95da68989ef3736c64ce1eaa4ef623b3a2f119c
Successfully built UNKNOWN
Installing collected packages: UNKNOWN
Successfully installed UNKNOWN-0.0.0
```

atinfinity, 2025-02-08:
https://github.com/pf-robotics/kachaka-api/issues/159#issuecomment-2642151057 で書いた

- Ubuntu 22.04
- Python 3.10

にてソースから入れることができない件ですが、Ubuntuインストール時点のpipバージョンが22.0であるため、pyproject.tomlベースでのインストールができないためでした（[参考](https://stackoverflow.com/a/78034907)）。

そのため、`pip install --upgrade pip`でpip自身をアップデートすればインストールできることを確認しました。

---

```
pip install --extra-index-url https://pf-robotics.github.io/kachaka-python-packages/simple kachaka-api
```

でインストールできない問題も`pip install --upgrade pip`でpip自身をアップデートすればインストールできることを確認しました。READMEにある情報だと気付けないので`pip install --upgrade pip`の手順を書くか、pipバージョン要件を書くのがよさそうです。

watanabe-pfr, 2025-02-09:
Ubuntu 22.04のaptで入る標準のpip (python3-pip) はversion 22.0.2で、このバージョンでは[PEP 621](https://peps.python.org/pep-0621/) が未サポートなのでsetup.pyがないと(pyproject.tomlだけでは)インストールできない状況でした。

カチャカのPlaygroundもUbuntu 22.04なので、手順の修正ではなく公開パッケージにsetup.pyを置くことにしました。既にsetup.pyがある状態のパッケージを公開済みです。

手元のx86_64環境でUbuntu 22.04のDockerイメージ内でインストールできるようになったことを確認しました。

ご報告ありがとうございました。

closed 2025-02-09 by watanabe-pfr