pf-robotics/kachaka-api #160: カチャカROS 2ブリッジでフレーム間のtransformができない

atinfinity, 2025-02-10:
## 現象

カチャカROS 2ブリッジでフレーム間のtransformができないため、例えば、RViz2でmapフレーム基準で可視化した際、LiDARデータが可視化できない。

## 環境

- Ubuntu 22.04
- ROS 2 Humble
- kachaka-api v3.8.5

## 再現手順

RViz2を起動する。

```bash
rviz2 -d ~/kachaka-api/ros2/demos/kachaka_nav2_bringup/rviz/kachaka-nav.rviz
```

カチャカROS 2ブリッジを起動する。

```bash
cd ~/kachaka-api/tools/ros2_bridge
./start_bridge.sh <カチャカのIPアドレス>
```

## 調査結果

<https://github.com/pf-robotics/kachaka-api/pull/137>で`frame_prefix`が導入されています。また、`frame_prefix`のデフォルト値が`/`になっています。その結果、フレーム名が`/map`、`/laser_frame`というように先頭に`/`が付くようになっています。そのため、tf2でフレーム間のtransformができなくなっています（[参考](https://b.ueda.tech/?post=20190629#ros%E3%81%AEtf2%E3%81%A7%E5%BE%AE%E5%A6%99%E3%81%AA%E4%BD%95%E3%81%8B%E3%81%8C%E3%81%82%E3%81%A3%E3%81%9F)）

上記のことから<https://github.com/pf-robotics/kachaka-api/pull/137>に伴うデグレである可能性が高いと考えています。

MrKeith99, 2025-02-12:
こちらの件に関しては，確認します．
確かに，namespaceを利用しない場合は，`/`をtfに入力しない方が良いです．

terakoji-pfr, 2025-02-12:
手元でビルドしたros2_bridgeを使うと発生しますね。こちらで修正します。

terakoji-pfr, 2025-02-13:
修正しました。新しいmeshも使われるようになります。

terakoji-pfr, 2025-02-13:
[image]

closed 2025-02-14 by terakoji-pfr