pf-robotics/kachaka-api #170: カチャカROS 2ブリッジでwheel_odometryとodometryに同一フレーム名が設定される

atinfinity, 2025-03-05:
## 現象

カチャカROS 2ブリッジにて`/kachaka/odometry/odometry`、`/kachaka/wheel_odometry/wheel_odometry`トピックに対して同じフレーム名が設定されています。例えば自己位置推定でnav2_amclを使う場合、<https://github.com/ros-navigation/navigation2/blob/1.3.5/nav2_amcl/src/amcl_node.cpp#L428-L438>にあるようにフレーム名でtf2のlookupTransformを行っているため、wheel_odometryとodometryが区別されずに混ざってしまうことが考えられます。こちらカチャカROS 2ブリッジの意図通りの振る舞いでしょうか？

## 環境

- Ubuntu 22.04
- ROS 2 Humble
- kachaka-api v3.8.5

## 再現手順

```bash
cd ~/kachaka-api/tools/ros2_bridge
./start_bridge.sh <カチャカのIPアドレス>
```
## 調査

以下のソースコードにて`/kachaka/odometry/odometry`、`/kachaka/wheel_odometry/wheel_odometry`トピックに対して同じフレーム名を設定しています。

- wheel_odometry：<https://github.com/pf-robotics/kachaka-api/blob/v3.9.5/ros2/kachaka_grpc_ros2_bridge/src/component/wheel_odometry_component.cpp>
- odometry：<https://github.com/pf-robotics/kachaka-api/blob/v3.9.5/ros2/kachaka_grpc_ros2_bridge/src/component/odometry_component.cpp>

CHangeLogを読むと<https://github.com/pf-robotics/kachaka-api/pull/119>に

> ros2_bridgeでwheel_odometryとodometryを同時に取得できない問題を修正。

とあるのでこの件のエンバグであると考えています。

terakoji-pfr, 2025-03-05:
ありがとうございます。
ここは同じ物理的座標系のフレーム座標系からフュージョンオドメトリと、純粋なホイールオドメトリがでているというのとなので問題無いと考えます。

同じカメラフレームから画像や認識結果がでているのと同じイメージです。

具体的な困りごとがありましたらお伝えいただけるとたすかります。

terakoji-pfr, 2025-03-05:
同じchild_frame_idのtf出してしまうと大問題ですが。

atinfinity, 2025-03-06:
@terakoji-pfr 説明ありがとうございます。勘違いしており申し訳ありません。tfトピックで出ている情報はフュージョンオドメトリによるものでしょうか？