pf-robotics/kachaka-api #172: tools/ros2_bridge/start_bridge.shのオプション処理がおかしい

atinfinity, 2025-03-12:
## 現象

`tools/ros2_bridge/start_bridge.sh`のオプション処理がおかしい

## 環境

- Ubuntu 22.04
- kachaka-api v3.9.5

## 説明

<https://github.com/pf-robotics/kachaka-api/blob/v3.9.5/tools/ros2_bridge/start_bridge.sh>のusageでは以下のような記述となっています。

```bash
Usage: ./start_bridge.sh KACHAKA_IP_ADRESS [KACHAKA_NAME] [Option]
  -d    daemonize
```

一方でstart_bridge.shに以下の記述があります。

```
if command -v docker-compose; then
    API_GRPC_BRIDGE_SERVER_URI="${KACHAKA_IP}:${GRPC_PORT}" NAMESPACE="${NAMESPACE}" FRAME_PREFIX="${FRAME_PREFIX}" docker-compose up "${@:4}" ros2_bridge 
else
    API_GRPC_BRIDGE_SERVER_URI="${KACHAKA_IP}:${GRPC_PORT}" NAMESPACE="${NAMESPACE}" FRAME_PREFIX="${FRAME_PREFIX}" docker compose up "${@:4}" ros2_bridge
fi
```

この処理の`${@:4}`は`[Option]`を渡すことを想定されているように思えますが、[Option]は3つ目以降のオプションであるため、`${@:3}`が正しいのではないかと考えています。おそらく<https://github.com/pf-robotics/kachaka-api/pull/137>の対応時にスクリプト引数は1つしか増えていないのに2つ増やしてしまっているのが問題だと考えています。

watanabe-pfr, 2025-03-13:
ご指摘ありがとうございます。確認して修正いたします。

また、kachaka-apiはオープンソースとして運営しております。修正案をお持ちでしたら、ぜひPull Requestとしてご提案いただけると大変助かります。

atinfinity, 2025-03-13:
@watanabe-pfr コメントありがとうございます。　https://github.com/pf-robotics/kachaka-api/pull/137 は多くのレビュアーの目が通されていることと、私が内部の作りや設計思想に詳しいわけではないで認識が合っているのかを確認させて頂きました。

PR #174 (2025-03-17): fixed the handling of argument in start_bridge.sh
I resolved #172
commit ba320a70 (2025-03-17): fixed the handling of argument in start_bridge.sh (#174)

I resolved #172