pf-robotics/kachaka-api #173: カチャカROS 2ブリッジで得られるodometryのcoverianceの値が非常に大きい

MizuhoAOKI, 2025-03-12:
## 現象
カチャカ公式の自己位置推定機能を利用しつつ走行中に, カチャカROS 2ブリッジにて`/kachaka/odometry/odometry`、`/kachaka/wheel_odometry/wheel_odometry`トピックを見ると, coverianceに大変大きな, 非現実的な値が入っています.　下記はある時刻のcoverianceの一例ですが, 値は毎回大きく異なります. 

```
 covariance:
  - 1970436014449635.2
  - 1139502937245978.8
  - 0.0
  - 0.0
  - 0.0
  - -1.3988636651251322e-18
  - 1139502937246009.5
  - 682264610546769.2
  - 0.0
  - 0.0
  - 0.0
  - 5.831626666289766e-19
  - 0.0
  - 0.0
  - 4.991703205090275e-07
  - 2.401840590378573e-18
  - -1.2470480311224804e-15
  - 0.0
  - 0.0
  - 0.0
  - 2.4018405903785733e-18
  - 4.983461230298931e-07
  - 3.610142062595648e-24
  - 0.0
  - 0.0
  - 0.0
  - -1.2470480311224808e-15
  - 3.610139100232841e-24
  - 4.983461230298911e-07
  - 0.0
  - -1.3988636651251324e-18
  - 5.831626666289767e-19
  - 0.0
  - 0.0
  - 0.0
  - 9.999966702581152e-10
```

なお, このとき, odometryの推定値(`pose.pose`)自体は, RViz上で可視化すると精度の良い値が推論できています.

## 再現手順
```
cd ~/kachaka-api/tools/ros2_bridge
./start_bridge.sh <カチャカのIPアドレス>
```

## ご質問
- 上記のcoverianceの値はどう解釈すればよろしいでしょうか？
  実用上, 特にx, y方向や車両の角度に関するcoverianceを得られると大変助かります.

以上, よろしくお願いいたします.

inabas-pfr, 2025-04-01:
現在、社内にて確認・調整を進めております。

MizuhoAOKI, 2025-04-01:
ご対応いただきありがとうございます.